openDAQ/openDAQ #170: Cannot build openDaq

Promar1, 2024-02-28:
Hi,

help wanted to build openDaq on my machine.
It seems like the parser is not getting set when I am trying to build the source code.
I am using Visual Studio 2019 for building.

There are several errors like the below error message:

**46>CUSTOMBUILD : warning : Could not determine if RTGen.Delphi.dll is a valid plugin.
45>CUSTOMBUILD : warning : Could not determine if RTGen.Delphi.dll is a valid plugin.
46>CUSTOMBUILD : warning : Could not determine if RTGen.Python.dll is a valid plugin.
45>CUSTOMBUILD : warning : Could not determine if RTGen.Python.dll is a valid plugin.**
**45>Could not find a parser for input language: cpp**
45>Known parsers:
**46>Could not find a parser for input language: cpp**
46>Known parsers:
46>C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): **error MSB8066: Custom build for** 'D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq\logger_sink_base_private.h;D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq\logger_sink_last_message_private.h;D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq\logger.h;D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq\logger_component.h;D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq\logger_sink.h;D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq\logger_thread_pool.h;D:\openDAQ-main\openDAQ-main\core\opendaq\logger\src\CMakeLists.txt' **exited with code 3.
46>Done building project "daq_logger.vcxproj" -- FAILED.**
38>ua_subscription_events_filter.c
55>------ Build started: Project: daq_context, Configuration: Debug x64 ------
45>C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): error MSB8066: Custom build for 'D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\property_object_internal.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\validator.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\coercer.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\component_type.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\property.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\property_internal.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\property_builder.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\property_object_protected.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\property_object_class.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\property_object_class_builder.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\property_object.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\ownable.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\eval_value.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\property_value_event_args.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\end_update_event_args.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\callable_info.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\argument_info.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\unit.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\unit_builder.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\include\coreobjects\core_event_args.h;D:\openDAQ-main\openDAQ-main\core\coreobjects\src\CMakeLists.txt' exited with code 3.
45>Done building project "daqcoreobjects.vcxproj" -- FAILED.

63>Building Custom Rule D:/openDAQ-main/openDAQ-main/build/__external/src/Boost/libs/locale/CMakeLists.txt
**58>D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq/logger_component_impl.h(19,10): fatal error C1083: Cannot open include file: 'opendaq/logger_thread_pool_ptr.h': No such file or directory**
58>D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq/logger_component_impl.h(19,10): fatal error C1083: #include <opendaq/logger_thread_pool_ptr.h>
58>D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq/logger_component_impl.h(19,10): fatal error C1083:          ^ (compiling source file D:\openDAQ-main\openDAQ-main\core\opendaq\logger\src\logger_component_impl.cpp)
58>D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq/logger_impl.h(19,10): fatal error C1083: Cannot open include file: 'opendaq/logger_sink_ptr.h': No such file or directory
58>D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq/logger_impl.h(19,10): fatal error C1083: #include <opendaq/logger_sink_ptr.h>
58>D:\openDAQ-main\openDAQ-main\core\opendaq\logger\include\opendaq/logger_impl.h(19,10): fatal error C1083:          ^ (compiling source file D:\openDAQ-main\openDAQ-main\core\opendaq\logger\src\logger_impl.cpp)

JakaMohorkoDS, 2024-02-28:
Hello @Promar1, have you made sure to fetch the LFS objects when cloning the repository?

Promar1, 2024-02-28:
Thanks for your reply, really appreciate that!
I have downloaded the source code zip file from github and followed instructions to build it.

Shall I clone the repository and then build the source code?

JakaMohorkoDS, 2024-02-29:
Thank you for the feedback. I do not think it should be an issue with downloading the source code as .zip. 

To give a bit of context of what is happening - openDAQ uses the "rtgen" tool to generate wrappers for C++  pointers, as well as Python, dotNET and Delphi bindings. The C++ generation is done as a custom build step, where it is failing in your case. 

Considering, the executable runs properly, my first guess would be that it might not have an adequate set of permissions when it tries to load the shared libraries (parser plugins) and is thus failing. Do you have a custom permission set/antivirus system that might be blocking the executable?

Promar1, 2024-02-29:
Thanks for your inputs!
The build issue got resolved after cloning the repository on my machine.
Not sure why zipped source code didn't work but cloned did.

I think Windows might have marked Zip file as a threat whereas git cloned version to be safe.

JakaMohorkoDS, 2024-02-29:
Glad to hear that you were able to get it to work!

closed 2024-02-29 by JakaMohorkoDS