openDAQ/openDAQ #383: Python bindings for armv8 (Jetson Xavier NX)

dennisappel, 2024-07-12:
Is it possible that you provide also python bindings for armv8 (Jetson Xavier NX)?

mojca, 2024-07-12:
It seems that GitHub finally supports ARM since last month:
https://github.blog/2024-06-03-arm64-on-github-actions-powering-faster-more-efficient-build-systems/

They are not available for open source projects before the end of the year though.

JakaMohorkoDS, 2024-07-15:
We're in the process of developing a deployment mechanism for ARM binaries for the SDK that should be available in upcoming months. Unless there's additional issues, it should be possible to generate Python bindings at that point as well. 

As of now, you would need to build the bindings on your own for the target platform - this can be done by enabling the `OPENDAQ_GENERATE_PYTHON_BINDINGS` CMake flag.

JakaMohorkoDS, 2024-07-18:
> I tried to build and got an .whl but it doesn't work. I will document the whole process and the result, when I have a little time. Thanks for your help. I will update here soon.

We'll also keep you posted on our progress with the binaries. While in draft stages for now, this PR should automate deployment for our ARM binaries https://github.com/openDAQ/openDAQ/pull/387

dennisappel, 2024-07-20:
To create the python bindings on arm (nvidia Jetson Xavier NX) I followed the instructions described in the readme and had to install some extra packages:
```shell
apt install -y git build-essential lld cmake ninja-build mono-complete python3
python3 -m pip install -U setuptools wheel pybind11 pybind11-stubgen numpy
``` 
After that:
```shell
cd build
cmake -DOPENDAQ_GENERATE_PYTHON_BINDINGS=ON ..
chmod +x ../bindings/python/run_rtgen.sh 
../bindings/python/run_rtgen.sh
``` 
While building the wheel, the "Wheel Tag:" was empty but the wheel was created: opendaq-3.0.0-cp310-cp310-.whl
I had to inject linux_aarch64 into the name. Also there was a bug with the METADATA for the numpy package, so I extracted the wheel changed the line (added 0) and packed it again:
```shell
numpy requires: >= 1.7.0, <=2.
changed to
numpy requires: >= 1.7.0, <=2.0
```
After that I was able to install the wheel, and it somehow worked but it seems there are problems which I can't figure out.
When I create an "opendaq.Instance()" I can loop over all available devices but only two ref devices show up.
When I run the same script with the precompiled wheel on an amd64 machine I get a lot of logs, the two ref devices and the opendaq machine which is in the network.

Tried to describe it as accurate as possible (from memory).

JakaMohorkoDS, 2024-07-22:
Good to hear you were able to build the bindings! As for only the reference devices being visible - openDAQ operates on a ["Module"](https://opendaq.github.io/opendaq/3.1.0/knowledge_base/modules.html) model where the core libraries contain a manager that dynamically loads modules during runtime.

These modules provide different openDAQ implementations of devices, function blocks, and server/client implementations. Did you use any preset to build the SDK? It could be that you did not compile the "Native Streaming", "LT Streaming", and "OPC UA" modules - all of these contain client implementations that can be used to connect to an openDAQ device on the network that runs a corresponding server.

If those are built and loaded, I'd check whether you can ping the openDAQ device on the network from your ARM device. `getAvailableDevices` runs an mDNS discovery query that should find all openDAQ services of devices on the network.

dennisappel, 2024-07-22:
I didn't use any presets, since no preset was designed for arm.
If "compile the "Native Streaming", "LT Streaming", and "OPC UA" modules" require additional flags I can try it, when the time allows it. Unfortunately we only have one device of that kind and the compilation takes around 5h without extra modules.
Can you list the flags that might solve the problem? I will try to compile it again.

JakaMohorkoDS, 2024-07-22:
Here are some relevant flags and descriptions that you can enable/disable depending on what you need.

```
// Reference implementations
DAQMODULES_REF_FB_MODULE           -> Generates example modules that provide function blocks such as the renderer and statistics
DAQMODULES_REF_DEVICE_MODULE       -> Generates a reference device implementation that outputs simulated data
DAQMODULES_AUDIO_DEVICE_MODULE     -> Example module that provides a device implementation using your device's audio card

// Server/client modules
DAQMODULES_OPENDAQ_CLIENT_MODULE   -> If enabled, client parts of server/client modules will be generated
DAQMODULES_OPENDAQ_SERVER_MODULE   -> If enabled, server parts of server/client modules will be generated
OPENDAQ_ENABLE_OPCUA               -> Generates OPC UA server/client modules
OPENDAQ_ENABLE_WEBSOCKET_STREAMING -> Generates Streaming LT server/client modules
OPENDAQ_ENABLE_NATIVE_STREAMING    -> Generates Native server/client modules

//  Misc
OPENDAQ_ENABLE_TESTS               -> True by default; Can be disabled if you do not require running tests
OPENDAQ_GENERATE_PYTHON_BINDINGS   -> Generates python bindings
```

The server/client modules are likely what you're missing. You need the client part to be able to discover and connect-to an openDAQ device running a corresponding server. The standard openDAQ simulator runs the Native Streaming and OPC UA servers. The former being the one you should be using if available, as it's the preferred method of connecting two openDAQ instances.

JakaMohorkoDS, 2024-08-07:
Issue closed due to inactivity.

closed 2024-08-07 by JakaMohorkoDS

dennisappel, 2024-08-15:
I am sorry to respond that late. I tried as suggested but wasn't able to communicate withe the device. Since we have a deadline we have to use an amd64 device so we can use the precompiled version you offer.
Thanks for your support and sorry that I can't give positive feedback.
Kind regards,
Dennis