openDAQ/openDAQ #409: Not getting OPCUA info while running cpp examples

irkahcp, 2024-07-26:
Hello Team,
I successfully ran quick_start example. But when I run the quick_start_full.exe I'm **not getting OPCUA deivces info**
[image]

While **executing the quick_start_full.exe** **I've noticed some warnings**  ( file attached )
[WarningLog.txt](https://github.com/user-attachments/files/16388993/WarningLog.txt)
[image]

Version I used:
           Open daq  version :  
	[opendaq-3.3.0-win64_f1631c8.exe](https://docs-dev.opendaq.com/download/SDK/opendaq-3.3.0-win64_f1631c8.exe)
Simulator version :
	[opendaq-3.3.0_device_simulator_f057af0.ova](https://docs-dev.opendaq.com/download/Simulator/opendaq-3.3.0_device_simulator_f057af0.ova)
	
I tried in Opendaq version of 
	[opendaq-3.0.0-win64.exe](https://docs.opendaq.com/download/SDK/opendaq-3.0.0-win64.exe)
Simulator of version : 
	[opendaq-3.0.0_device_simulator.ova](https://docs.opendaq.com/download/Simulator/opendaq-3.0.0_device_simulator.ova)
in this case also I'm not able to get any opcua device information 

One more thing I need to add.
when I used opendaq-2.0.0-win64 and opendaq-2.0.0_device_simulator last time, I was able to get opcua  device info  like
        Device Name : Device 1
        string getconnection : daq.opcua://191.168.56.101.7420 

[image]

can you please assist in this. any help would be really appreciated

Thank you.

JakaMohorkoDS, 2024-07-30:
Hello @irkahcp, 

The changes you are seeing are a result of a change in the openDAQ device discovery system. A device might have multiple servers running - in particular the simulator starts up the [Native and OPC UA servers](https://opendaq.github.io/opendaq/3.3.0/knowledge_base/protocol_features_characteristics.html). In discovery, a device with multiple servers will now appear only once, having its server information grouped within the device info list of ["Server Capabilities"](https://opendaq.github.io/opendaq/3.3.0/howto_guides/howto_connect_to_device.html).

We are working on providing more detailed documentation on these topics, but the recent iteration of the [Quick Start guide](https://opendaq.github.io/opendaq/3.3.0/getting_started/quick_start_application.html#_discovering_devices) also highlights this change and explains the behaviour.

Hope this answers your questions!

irkahcp, 2024-08-05:
Thanks for your help. now I'm able to connect desired server.

closed 2024-08-05 by JakaMohorkoDS