openDAQ/openDAQ #426: Websocket device name

gallevatohbk, 2024-08-01:
Websocket device names are hardcoded to 
'WebsocketClientPseudoDevice'
("websocket_client_device_impl.cpp)

The streaming protocol has no meta option to set the device.info.name.

JakaMohorkoDS, 2024-08-07:
Hello, 

In general, the device info can never be changed. It's information about the device that's read-only. I believe we do get some information on the device's name via the LT protocol that could be used in the device's info, as well as in the component's name, but I'm not 100% sure on that part. @NikolaiShipilov Is there any information we could use from LT to get a more appropriate device name?

gallevatohbk, 2024-08-07:
Hi,

maybe the device.info.name could be set by using the same name of the MDNS TXT response.
This would be the device name which is already displayed when using the openDAQ discovery

NikolaiShipilov, 2024-08-07:
There is no dedicated meta-data field in the LT protocol for passing the device name; Only the names and IDs of specific signals are available.

JakaMohorkoDS, 2024-08-07:
> Hi,
> 
> maybe the device.info.name could be set by using the same name of the MDNS TXT response. This would be the device name which is already displayed when using the openDAQ discovery

The mDNS name is not always available. Indeed, if discovery information is available, we could use that instead, but that leads to somewhat tricky logic:

1. If `instance.getAvailableDevices()` is called before `instance.addDevice()`, the discovery name would be used
2. Otherwise, the "pseudo device" name is used.

Thinking about it, this could be implemented in a manner where we would always run discovery before adding a device. In that case, as long as the device is discoverable, it would have an appropriate name. It would, however, lengthen the `addDevice` call by 1-2s.

I propose a solution where:
1. The discovery name is used in LT streaming devices
2. Discovery is run on `addDevice`, but can be disabled via a `General` setting in the `addDeviceConfig` object.
3. I'd change the names of current "pseudo" devices to "streaming" devices. Eg. "LTStreamingDevice". Similarly, this should be implemented for the native devices

There's a question on whether we should scan for devices if the discovery info has already been cached in the module manager. I'd say we should not, but it might be worth storing a timestamp on the discovery data, and re-scanning on `addDevice` if the scan information is, let's say, >10-30s old.

@dejancrn @NikolaiShipilov Thoughts?

NikolaiShipilov, 2024-08-07:
Wouldn't including the device name in the LT protocol 'init' method parameters be a more straightforward way to pass it?
Additionally, a similar addition could be made for the native streaming protocol.

JakaMohorkoDS, 2024-08-08:
> Wouldn't including the device name in the LT protocol 'init' method parameters be a more straightforward way to pass it? Additionally, a similar addition could be made for the native streaming protocol.

This implies a modification to the protocol. I'd add a fix that does not need to change anything, and we're already using the discovered device data to build up the `DeviceInfo` (server capabilities). I don't see this as too big of a change. Also - exactly due to the server capabilities, I anyhow see scanning before `add` as a good addition.