openDAQ/openDAQ #463: Accessing OPC UA library methods  in openDAQ.

irkahcp, 2024-08-22:
Hello team,

Earlier I used OPCUA protocols to communaicate with my devices. I'm going to replacing the OPCUA library with openDAQ library and I need some clarification,  whether its possible to use same opcua calls in openDAQ sdk or not 
Scenario :
#define DEWESOFT_STRING_FIRST_CHANNEL_NAME "data/ Daq Module/channels/SIRIUS_BLACK-XHS 0/Analog inputs/AI 1_Name"
 const CHAR* m_pcChannelNameName           = DEWESOFT_STRING_FIRST_CHANNEL_NAME;
 
 Im passing this m_pcChannelNameName varialble to SetConfigurationDataChar() method inorder to change configuration of device.( refer below image )
 
[image]

 In SetConfigurationDataChar() method, based on m_pcChannelNameName OpcUaNodeId was created and by using this node we are configure the device by calling one of the opcua62541 lib method UA_Client_writeValueAttribute(UA_Client *client, const UA_NodeId nodeId,const UA_Variant *newValue) ( refer below images )
 
[image]

[image]

Now my concern is, in openDAQ documentation it's mentioned that  **openDAQ™ extends the standard with its own node sets and modeling rules built upon the [OPC 10000-100: Devices node set]** ( refer below image )
[image]

as openDAQ **extends the standard with its own node sets and modeling rules built upon the [OPC 10000-100: Devices node set],** how can we configure the device by using OpcUaNodeId in openDAQ  ? means is it possible to use OpcUaNodeId ( OPCUA Nodes ) in openDAQ  inorder to configure the device? or any other way is there to configure the device in openDAQ, like using component structure ( eg:  Device -> Channel -> Properties -> Name ). 

Can you please assist in this, any help would be extremely helpful for me.

Thank you.

JakaMohorkoDS, 2024-08-27:
I'm not exactly sure on the use-case in the above examples. openDAQ is based on a set of OPC UA modelling rules that are used to publish the device's structure on an OPC UA server. 

If an OPC UA server is started on an openDAQ device, you can connect to it with any generic OPC UA client (eg. UA Expert). You can call any standard OPC UA methods on the device. A thing to keep in mind is that unless you have a client that parses nodesets during runtime (open62541 does not), you will need to import the openDAQ OPC UA nodeset that can be found in the [openDAQ OPC UA companion spec repository](https://github.com/openDAQ/opc-ua-companion-spec)

Notably, you cannot connect with openDAQ to any OPC UA server - openDAQ does not have a generic OPC UA client, the server must match the modelling rules imposed. Non-openDAQ OPC UA devices (eg. older Dewesoft devices) likely will not follow said rules and you will not be able to connect to them with an openDAQ client.

If you have any further questions, or I misunderstood the question, please let me know!

irkahcp, 2024-08-28:
Thanks for your response.
As you mentioned, as openDAQ does not have generic OPC UA cleint, it's not possible to use OPC UA methods through openDAQ and In openDAQ devices once OPC UA server starts then we can call OPC UA methods through OPC UA client. please correct me if  I wrong.
so my query here is , once we  import the openDAQ OPC UA nodeset is it possible to use both OPC UA methods and openDAQ methods simultaneously?. As I'm planning to replace OpcUA library with openDAQ, is it possible use both opcua methods and openDAQ methods simultaneously with one instance or node?
The reason behind this question is that I previously used OPC UA methods. If I can replace OPC UA with openDAQ and still use the previously implemented methods in openDAQ, then I won't need to implement them from scratch. This would also allow me to take advantage of any new features available in openDAQ. So is there any way to use previous opcua methods in opendaq ?

please let me know if you had any confusion.

JakaMohorkoDS, 2024-08-30:
openDAQ provides a toolkit in the form of modules that abstracts away the transport layer - basically in the case of the OPC UA client module, it wraps the open62541 calls and calls them when a corresponding SDK method is called. You can completely circumvent that by doing the OPC UA calls manually - so your current approach will work, but it will be harder to implement.

Notably, the OPC UA model is very specific to openDAQ - unless you were generically parsing the OPC UA tree previously, you might find that your approach of parsing device OPC UA structures is no longer compatible. I'd suggest comparing the OPC UA tree structure of openDAQ to the OPC UA structure of the device you're currently working with using a generic OPC UA client such as UA Expert.

Hope this answers your question.

closed 2024-09-24 by JakaMohorkoDS