openDAQ/openDAQ #499: Can't install openDAQ after build

white5354, 2024-09-12:
When i Building openDAQ™ from source.When after execute `cmake --build .` I meet a problem when i execute `cmake --install . --prefix "openDAQ"`.
 
Below 
[image]

And I find my file no **Release** file. only Debug file.

JakaMohorkoDS, 2024-09-13:
Hello, it seems that fmt always tries to install the release target, but the SDK was built in debug. I'd suggest building the SDK in release mode by calling `cmake --build . --config Release`. 

Explicitly specifying the config when installing seems to also work: `cmake --install . --prefix "openDAQ" --config Debug`. The documentation should be updated to clearly reflect the commands.

JakaMohorkoDS, 2024-10-01:
Addressed in https://github.com/openDAQ/openDAQ/pull/522

closed 2024-11-01 by JakaMohorkoDS