openDAQ/openDAQ #504: No find Devices

white5354, 2024-09-13:
Hello, when I follow the documentation [openDAQ™ client application](https://docs.opendaq.com/manual/opendaq/3.0.0/getting_started/quick_start_application.html#own_simulator),I start Virtual box simulator image,below this
[image]

but I note this 
```
import opendaq

# Create an openDAQ(TM) instance
instance = opendaq.Instance()

# Find and output the names and connection strings of all available devices
for device_info in instance.available_devices:
    print("Name:", device_info.name, "Connection string:", device_info.connection_string)
    for capability in device_info.server_capabilities:
        print("Capability:", capability.protocol_name, "Connection string:", capability.connection_string)
```
There are no devices,look below
[image]

What did I do wrong?

JakaMohorkoDS, 2024-09-13:
Note quite sure what you mean by "there are no devices," as the output shows that there are 3 devices found. The "Reference device simulator" has 3 server capabilities, while 2 more devices are simulators that you can add locally (Device 0 and Device 1).

white5354, 2024-09-14:
@JakaMohorkoDS thx for your reply.Your are right.I just want to get the exports like 
[image]
maybe i should believe in code.

closed 2024-09-14 by white5354