openDAQ/openDAQ #555: [BUG] <Configuration loading through JSOn not working >

rahul19861986, 2024-10-18:
### Current behavior

loadConfiguration(json); is not working, its not updating the previous configurations, but function returns 0

### Expected behavior

loadConfiguration(json); should update the new values in JSON or return an error 

### Code or steps

```
<loadConfiguration(json); with this JSON file 
{
    "__type": "Instance",
    "rootDevice": {
        "a24433da-14a7-4acb-8c9b-a2efb2b483af": {
            "__type": "Device",
            "name": "openDAQ Client",
            "Dev": {
                "__type": "Folder",
                "items": {
                    "ref_dev1": {
                        "__type": "Device",
                        "IO": {
                            "__type": "IoFolder",
                            "items": {
                                "ai": {
                                    "__type": "IoFolder",
                                    "items": {
                                        "refch0": {
                                            "__type": "Channel",
                                            "typeId": "ref_channel",
                                            "Sig": {
                                                "__type": "Folder",
                                                "items": {
                                                    "ai0": {
                                                        "__type": "Signal",
                                                        "domainSignalId": "/a24433da-14a7-4acb-8c9b-a2efb2b483af/Dev/ref_dev1/IO/ai/refch0/Sig/ai0_time",
                                                        "dataDescriptor": {
                                                            "__type": "DataDescriptor",
                                                            "name": "AI 1",
                                                            "sampleType": 2,
                                                            "unit": {
                                                                "__type": "Unit",
                                                                "symbol": "V",
                                                                "name": "volts",
                                                                "quantity": "voltage"
                                                            },
                                                            "dimensions": [],
                                                            "valueRange": {
                                                                "__type": "Range",
                                                                "low": -10.0,
                                                                "high": 10.0
                                                            },
                                                            "rule": {
                                                                "__type": "DataRule",
                                                                "ruleType": 3,
                                                                "params": {
                                                                    "__type": "Dict",
                                                                    "values": []
                                                                }
                                                            },
                                                            "origin": "",
                                                            "metadata": {
                                                                "__type": "Dict",
                                                                "values": []
                                                            },
                                                            "structFields": []
                                                        },
                                                        "public": true
[... 365 lines cut ...]
                                            }
                                        }
                                    }
                                },
                                "can": {
                                    "__type": "IoFolder"
                                }
                            }
                        },
                        "sync": {
                            "__type": "Component"
                        }
                    }
                }
            }
        }
    }
}
>
>

### Environment

* openDAQ version tag or commit SHA: 3.000
* OS: windows

aljazfrancic, 2024-10-23:
Can you try this with a more recent version of openDAQ? For example: https://docs-dev.opendaq.com/
You can find detailed installation instructions for C++, C# and Python here: 
https://docs-dev.opendaq.com/manual/opendaq/3.4.0/getting_started/quick_start.html

aljazfrancic, 2024-10-23:
Also make sure to use the same version of openDAQ for both saving and loading the configuration.

rahul19861986, 2024-10-23:
Unfortunately, it's not working as expected. I tried building the latest version from the master branch, as there's no release tag for version 3.4, but it crashes with an error when I attempt to create an instance

[image]

denise-opendaq, 2024-10-24:
Hello @rahul19861986

I see that the module manager is trying to find the libraries in `/usr/local/lib/modules`. Are you sure the built modules are located in this path? If the folder path is different, you can either override the environment variable `OPENDAQ_MODULES_PATH` or replace `MODULE_PATH` with the actual path to the libraries in `examples/cpp/stream_reader_example/stream_reader_example.cpp`.

Now, regarding the initial problem 😉: The signal is not storing the descriptor because setting the descriptor is the responsibility of the signal's owner. I see that you're attempting to change the sample rate of the signal for the reference device. To modify the signal's delta, you should adjust the sample rate of the channel. To do this, set the channel property `UseGlobalSampleRate` to `false`, and then assign a custom value to the `SampleRate` property. Here's an example of how to do this:
```
int main(int /*argc*/, const char* /*argv*/[])
{
    StringPtr config;
    {
        // Create an Instance, loading modules at MODULE_PATH
        const InstancePtr instance = Instance(MODULE_PATH);
        auto device = instance.addDevice("daqref://device0");
        std::cout << "Initial configuration:" << std::endl;
        std::cout << instance.saveConfiguration() << std::endl;
        
        auto channel = device.getChannels()[0];
        channel.setPropertyValue("UseGlobalSampleRate", false);
        channel.setPropertyValue("SampleRate", 200.0);
        config = instance.saveConfiguration();
        std::cout << "Modified configuration with new sample rate:" << std::endl;
        std::cout << config << std::endl;
    }

    const InstancePtr instance = Instance(MODULE_PATH);
    instance.loadConfiguration(config);
    std::cout << "Loaded configuration:" << std::endl;
    std::cout << instance.saveConfiguration() << std::endl;
    return 0;
}
```

VladDoc, 2024-10-26:
I was able to load the configuration on 3.4, however I've found another issue:
If you load the configuration with a device specified, it locks up addition of that device(when creating an instance from scratch) until you restart the simulation VM.
Steps to reproduce:
 - boot up fresh VM
 - open up gui_demo.py from python examples
 - add first reference device to an instance
 - File -> Save configuration to config.json
 - close gui_demo.py
 - open gui_demo.py
 - File -> Load configuration from config.json (*everything loads properly*)
 - close gui_demo.py
 - open gui_demo.py
 - notice that no devices are present
 - add first reference device to an instance (*nothing happens*)
 - if you try to load configuration again device is also not added
 
Info I've found out during debugging:
 - `daq::GenericDevicePtr<daq::IDevice>::addDevice` method returns error code 0x80000025(OPENDAQ_ERR_DUPLICATEITEM)  suggesting that it fails due to device being a duplicate
 - However `device.getDevices().getCount()` returns 0 before and after the call so you cannot select a device either
 - `device.getAvailableDevices()` returns a list with this locked up device included

Console logs if you try to add the device after loading config previously
```
python3 gui_demo.py
[tid: 14854][2024-10-27 01:06:01.705] [ModuleManager] [warning] Failed to check for network settings: open: Operation not permitted [system:1 at /__w/openDAQ/openDAQ/build/__external/src/Boost/libs/asio/include/boost/asio/detail/reactive_socket_service.hpp:133 in function 'open']
Error adding device daq://openDAQ_08:00:27:16:5c:7f: Signal with Ids (global /// remote /// key) "/9a415a3b-3ff9-405f-adc1-2732cdfd96dd/Dev/opendaq_08:00:27:16:5c:7f/Dev/opendaq_08:00:27:16:5c:7f/IO/AI/RefCh0/Sig/AI0" /// "/opendaq_08:00:27:16:5c:7f/Dev/opendaq_08:00:27:16:5c:7f/IO/AI/RefCh0/Sig/AI0" /// "/opendaq_08:00:27:16:5c:7f/IO/AI/RefCh0/Sig/AI0" failed to add - signal already added to streaming "daq.ns://X.X.X.X:7420/"
```

Console logs if you try to load the configuration second time after the VM launch:
```
[tid: 14854][2024-10-27 01:11:07.620] [ModuleManager] [warning] Failed to check for network settings: open: Operation not permitted [system:1 at /__w/openDAQ/openDAQ/build/__external/src/Boost/libs/asio/include/boost/asio/detail/reactive_socket_service.hpp:133 in function 'open']
[tid: 14854][2024-10-27 01:11:07.977] [/9a415a3b-3ff9-405f-adc1-2732cdfd96dd] [warning] Failed to update device: Invalid parameter
```

VladDoc, 2024-12-12:
Tested save/load configuration on the latest release 3.10 and the issue does not reproduce with the repro steps I specified above

closed 2025-04-30 by denise-opendaq