openDAQ/openDAQ #571: [QUESTION] <How to form a custom daq device as openDAQ server>

hustfox, 2024-10-30:
### Question

I have a sound level meter device which can transfer audio data via tcp socket. I already have a software that can connect to the sound level meter and receive the audio data.  But i have no idea how to use the openDAQ sdk to collect the audio data. I need your help.

JakaMohorkoDS, 2024-10-30:
Hello @hustfox 

To integrate you device into openDAQ, you need to create a `Module` that provides the option to add your device to an openDAQ instance. I'd suggest looking at the "Audio Device Module" available in the "modules/audio_device_module" folder. 

Said implementation provides:
- An openDAQ device implementation that uses the `miniaudio` library to read audio card data
- A WAV file writer module that writes signal data into a .wav file

You can model your device similarly:
- in your `onCreateDevice` override, connect to your device using your device driver
- in `onGetAvailableDeviceTypes`, you should specify a connection string prefix (like `miniaudio://` that will be used to connect to your device in the type object you provide as the return value
- `onGetAvailableDevices` should should return a list of devices discovered on the network, if your device supports a discovery mechanism

Sadly we currently do not yet have guides available for either supporting devices or writing function blocks in openDAQ. We are working on providing those in upcoming releases.

Let me know if this helps!

hustfox, 2024-10-31:
Thanks for your reply. 
So the only way to add my own daq device to a openDAQ server is by implementing a custom miniaudio backend? I've consulted the Devesoft software developer for this question, they said they use a private openDAQ protocol to achieve this.

JakaMohorkoDS, 2024-11-01:
@hustfox @dodomouse 

There are currently no guides available on supporting a device in openDAQ - we only have examples. The two examples are the miniaudio device and the openDAQ reference device (ref_device_module).

Why I used the miniaudio device as an examples is that it highlights how an existing device driver is used to integrate a device into openDAQ. You do not need to start an openDAQ server on the device to make it openDAQ-compliant if you already have your own method of connecting to the device. 

You need to provide a module similar to that of the miniaudio device implementation that uses your driver to connect to the device, then translates its structure and data into that of openDAQ.

But to clarify, in general there are 3 ways of supporting a device in openDAQ:

1) The method outlined above, where no openDAQ server is started on the device. In this case you must distribute a module that uses your drivers to connect to the device.

2) Starting an openDAQ server on the device - viable for devices with a Linux/Windows/Mac OS. In this case you must provide the same module as outlined in 1), with the exception that you are running the driver on the hardware itself - of course in that case you acquire the data from a different API of your device. After writing the integration module, you can then start an openDAQ server on it as follows:

```
// Creates your device and sets it as the root device of the openDAQ tree
const auto instance = InstanceBuilder().setRootDevice("myprefix://mydevice").build();

// Creates an openDAQ server
instance.addServer("OpenDAQNativeStreaming");
```

To see an example of how this is done in our simulator, you can check the "examples/cpp/quick_start/quick_start_simulator.cpp" example.

3) Not using openDAQ on your device at all, but implement a server stack that matches the requirements of one of the openDAQ protocols (Native, OPC UA, LT streaming). This would be the most difficult approach, as the protocols themselves are not documented in detail. If you would wish to take this approach, however, documentation outlining all of the specs is available, and the source code can be used as an example implementation.

Notably, there is documentation available that highlights the openDAQ structure:
https://opendaq.github.io/opendaq/main/knowledge_base/opendaq_architecture.html

Additionally, some brief information on different openDAQ servers is available here:
https://opendaq.github.io/opendaq/main/knowledge_base/protocol_features_characteristics.html

How modules work is oulined in the following document:
https://opendaq.github.io/opendaq/main/knowledge_base/modules.html

Hopefully this helps out a bit and lets you delve a bit deeper into openDAQ. As mentioned above, we are working on providing additional information on module development, and a template for module development is being worked on.

dodomouse, 2024-11-01:
> @hustfox @dodomouse
>
> There are currently no guides available on supporting a device in openDAQ - we only have examples. The two examples are the miniaudio device and the openDAQ reference device [...]

Thanks a lot :)

JakaMohorkoDS, 2024-11-06:
Glad to be of help! Hope the above is sufficient to get you started.

closed 2024-11-06 by JakaMohorkoDS