openDAQ/openDAQ #677: [BUG] Can't change List Property remotely with Python GUI

sedovmo, 2025-01-23:
### Current behavior

I have an Opedaq server and a Function block with a List property instantiated on the server. I can't change this List property remotely using the Python GUI. I get "Can't add item: Unsupported core type: CoreType.ctUndefined" error.

[image]

But I can change this property using C++, so it seems like the Python only issue

### Expected behavior

I can change this property if I add the Function block locally.

[image]

### Code or steps

Run a device server. Open the Python GUI.
In Python GUI
1. Add device with Connection string "daq.nd://127.0.0.1"
2. Add Function Block with List property in it
3. Double click this List property, you'll see a new "Edit" window
4. Press "Add" button, you'll see a new "Add item" window
5. Type some text and press OK button
You'll see the error

### Environment

* openDAQ version tag or commit SHA:
Python GUI under Ubuntu -  Opendaq 3.11.0.d3f7343
Python GUI under Windows 11 - Opendaq 3.11.0.ae8716f

* OS: server - Ubuntu 22.04.5, python GUI run under Ubuntu 22.04.5  and Windows 11.
* programming language: Python

aljazfrancic, 2025-01-30:
Trying to reproduce this, I made a server like this:

```cpp
#include <iostream>
#include <opendaq/opendaq.h>

using namespace daq;

int main(int /*argc*/, const char* /*argv*/[])
{
    const InstancePtr instance = Instance("");
    instance.setRootDevice("daqref://device0");
    auto dummy = instance.addStandardServers();
    instance.addProperty(ListProperty("listProp", List<IBaseObject>("foo", "bar")));
    std::cin.get();
    return 0;
}
```
On the latest main.

Then I add device with Connection string `daq.nd://127.0.0.1` in Python GUI app.

This is somewhat different because it's on a Device instead of a Function Block.

This works fine, with either `IBaseObject` or `IString` as the list templated type.

However, if I, for example, do:

```cpp
instance.addProperty(ListProperty("listProp", List<IString>()));
```

(e.g. empty list of whatever type)

this will fail with the error you get as well.

Anyways, here's the fix:
https://github.com/openDAQ/openDAQ/pull/686

aljazfrancic, 2025-01-30:
@JakaMohorkoDS This could ve affecting Dictionaries as well. I will check and report back my findings.

sedovmo, 2025-01-30:
With this fix it works on my side as well (I can add items to an empty string list). Thank you!
Just one small thing is that in case of empty string list it shows CoreType.ctUndefined at the top of the window. But this doesn't affect the functionality.

[image]

aljazfrancic, 2025-01-30:
The core of the problem is that for List `self.property.item_type`, and for Dictionary `self.property.key_type`, `self.property.item_type` have erroneous values, e.g. `ctUndefined`. Dictionaries are also broken. I'll investigate further.

PR #733 (2025-03-18): Add element IntfID to list/dict serialization + Serialization versioning
# Brief

- Fixes list/dictionary deserialization not containing key/value/item interface IDs.
- Introduces serializer versioning; openDAQ list objects are now serialized as objects instead of JSON arrays.

# Description

List and Dictionary objects in openDAQ store a key/value/list interface ID internally that can be used to retrieve the expected interface of stored objects. This is set on object construction if specified as a constructor parameter.

However, the serialization/deserialization of objects did not store/rest [... cut]