openDAQ/openDAQ #688: [BUG] Critical segmentation fault

gallevatohbk, 2025-01-31:
### Current behavior

Hey guys,

I sporadically run into a segmentation fault after adding an (openDAQ LT) device and trying to start a reader for a signal due to the following issue:

After instance.add_device(...) I get the error message:

[tid: 14540][StreamingClient] [error] [SignalContainer.cpp:190] No time signal available for signal id 'BM60_CH3.value', number 7, table 'BM60_CH3'!
[tid: 14540][StreamingClient] [error] [ProtocolHandler.cpp:81] failed to interprete measured data!: protocol error

After that, printing out the signal names (device.get_signals_recursive()) and their domains for this device, the missing time signal is included:

Signal ID: 0 | Name: BM60_CH1.value | Domain: BM60_CH1.time
Signal ID: 1 | Name: BM60_CH2.value | Domain: BM60_CH2.time
**Signal ID: 2 | Name: BM60_CH3.value | Domain: BM60_CH3.time**
Signal ID: 3 | Name: BM60_CH4.value | Domain: BM60_CH4.time
Signal ID: 4 | Name: BM60_CH1.time
Signal ID: 5 | Name: BM60_CH1.status | Domain: BM60_CH1.time
Signal ID: 6 | Name: BM60_CH2.time
Signal ID: 7 | Name: BM60_CH2.status | Domain: BM60_CH2.time
**Signal ID: 8 | Name: BM60_CH3.time**
Signal ID: 9 | Name: BM60_CH3.status | Domain: BM60_CH3.time
Signal ID: 10 | Name: BM60_CH4.time
Signal ID: 11 | Name: BM60_CH4.status | Domain: BM60_CH4.time

but the error message states it is looking for number 7 somehow? I think it should be 8?

After that, trying to setup daq.StreamReader for the signals 0 to 3 results in

----- setup signals-----
[1]    12335 segmentation fault

Possibly, a similar segmenation fault is caused by the following sequence:

instance.add_device(..)

[tid: 6768][2025-01-31 15:00:50.990] [StreamingClient] [error] [SignalContainer.cpp:142] Got data for signal '7', that was not subscribed before
[tid: 6768][2025-01-31 15:00:50.990] [StreamingClient] [error] [ProtocolHandler.cpp:81] failed to interprete measured data!: protocol error
[tid: 15012][2025-01-31 15:00:51.990] [StreamingClient] [warning] signal BM60_CH3.value has incomplete descriptors
[tid: 15012][2025-01-31 15:00:51.990] [StreamingClient] [warning] signal BM60_CH4.value has incomplete descriptors

Printing the signals

Signal ID: 0 | Name: BM60_CH1.value | Domain: BM60_CH1.time
Signal ID: 1 | Name: BM60_CH2.value | Domain: BM60_CH2.time
**Signal ID: 2 | Name: BM60_CH3.value
Signal ID: 3 | Name: BM60_CH4.value**
Signal ID: 4 | Name: BM60_CH1.time
Signal ID: 5 | Name: BM60_CH1.status | Domain: BM60_CH1.time
Signal ID: 6 | Name: BM60_CH2.time
Signal ID: 7 | Name: BM60_CH2.status | Domain: BM60_CH2.time
Signal ID: 8 | Name: BM60_CH3.time
Signal ID: 9 | Name: BM60_CH3.status | Domain: BM60_CH3.time

CH3 and CH4 are missing the domain information, and these are missing completely:
Signal ID: 10 | Name: BM60_CH4.time
Signal ID: 11 | Name: BM60_CH4.status | Domain: BM60_CH4.time

Results in
[1]    18293 segmentation fault

Maybe there is correlation between those two errors.
They happen sporadically, but make the complete system unreliable, since it seems that the error can not be catched by try - except blocks in Python.
Is there any way to check for these errors in python and maybe repeat the add_device procedure?

Thanks in advance.

### Expected behavior

No segmentation faults when StreamReader are opened and all signals included:

Signal ID: 0 | Name: BM60_CH1.value | Domain: BM60_CH1.time
Signal ID: 1 | Name: BM60_CH2.value | Domain: BM60_CH2.time
Signal ID: 2 | Name: BM60_CH3.value | Domain: BM60_CH3.time
Signal ID: 3 | Name: BM60_CH4.value | Domain: BM60_CH4.time
Signal ID: 4 | Name: BM60_CH1.time
Signal ID: 5 | Name: BM60_CH1.status | Domain: BM60_CH1.time
Signal ID: 6 | Name: BM60_CH2.time
Signal ID: 7 | Name: BM60_CH2.status | Domain: BM60_CH2.time
Signal ID: 8 | Name: BM60_CH3.time
Signal ID: 9 | Name: BM60_CH3.status | Domain: BM60_CH3.time
Signal ID: 10 | Name: BM60_CH4.time
Signal ID: 11 | Name: BM60_CH4.status | Domain: BM60_CH4.time

### Code or steps

see above

### Environment

* openDAQ version tag or commit SHA: opendaq 3.10.3
* OS: Windows 11
* programming language: python 3.13.1

JakaMohorkoDS, 2025-02-05:
Hello @gallevatohbk would it be possible to get a description of the signals on the server side? What data they're streaming and what descriptors they have. This issue is not something we've observed on our side up to this point, and some further pointers on the server side implementation would be useful in our attempts to mimic your setup and reproduce the defect.

NikolaiShipilov, 2025-02-05:
Hello @gallevatohbk , 

Analysis of a previously reported similar bug suggests that the issue originates from the LT protocol message sequence not adhering to the expected flow, causing the LT connection to terminate abnormally. The requirement is that the meta-information for time and value signals must be received first, as the client relies on it to correctly interpret the value signal data.

Assuming that here, as with another bug report, the use case involves a simplified embedded version of the LT server rather than an openDAQ LT server implementation on the device side, the root cause of the issue likely needs to be addressed by the FW dev team within the device-server implementation (suggestions for this have already been shared in a separate discussion). 

However, to prevent abnormal termination of the LT connection session when a data processing failure occurs, modifications are planned on the client-side behavior, which should also resolve the segfault mentioned here.

NikolaiShipilov, 2025-02-11:
Hello @gallevatohbk ,

The client-side changes intended to address the segfault are now available for testing under the tag https://github.com/openDAQ/openDAQ/releases/tag/v3.10.11-dev

closed 2025-02-21 by gallevatohbk